iNavFlight/inav-rangefinder #4: 2.1.0-RC3 RANGEFINDER=UNAVAILABLE

Robotto, 2019-02-25:
Does not seem to work on.
# INAV/KAKUTEF4V2 2.1.0 Feb 10 2019 / 16:13:09 (0d3d0ff30)
Is there a specific firmware where this is confirmed working?

Robotto, 2019-03-08:
ohsnap, i didn't compile the firmware with the right config.

for future reference, I added the following to /src/main/target/KAKUTEF4/target.h:
```
#define USE_RANGEFINDER
#define RANGEFINDER_I2C_BUS     BUS_I2C1
#define USE_RANGEFINDER_HCSR04_I2C
#define RANGEFINDER_HCSR04_I2C_I2C_INSTANCE (I2C_DEVICE)
```
and the following to /src/main/target/KAKUTEF4/target.mk:

```
drivers/rangefinder/rangefinder_hcsr04.c \
drivers/rangefinder/rangefinder_hcsr04_i2c.c \
```

closed 2019-03-08 by Robotto

thijn1234, 2020-01-03:
@Robotto where do you add that?

Robotto, 2020-01-03:
> @Robotto where do you add that?

Here: https://github.com/iNavFlight/inav-rangefinder/issues/4#issuecomment-470858456

thijn1234, 2020-01-04:
I Kown but where do you at the 
/src/main/target/KAKUTEF4/target.h and the /src/main/target/KAKUTEF4/target.mk
?

Robotto, 2020-01-04:
I'm sorry I don't understand the question.

Have you read the build guide? It's here:
https://github.com/iNavFlight/inav/wiki/Building-custom-firmware

thijn1234, 2020-01-07:
thanks for helping but I have a problem 
[image]
do you have an answer?

Robotto, 2020-01-07:
Perhaps try 'make clean' before making?

thijn1234, 2020-01-07:
no that won't work

Robotto, 2020-01-07:
Okay.
It seems that the linker is acting up. Try manually deleting build output
files, and make sure you're on a branch that supports the kakutef7mini
target.

Robotto, 2020-01-07:
This discussion is turning into general troubleshooting.
May I suggest that you direct further questions to the inav group on
Telegram?

thijn1234, 2020-01-08:
oke I flash the new firmware version but what now

Robotto, 2020-01-08:
Now you fly.