iNavFlight/inav-rangefinder #6: Inav_I2C with US100 + Arduino nano only shows altitude up to 60cm

minhdungDIY, 2019-04-02:
Hi guys,
I successfully set up the range finder using a US 100 sensor and an arduino nano, connected to I2C bus. I had to manually included the HCSR04 driver to the target files (as mentioned in closed issues #4  and #5). Everything seems to work fine, the altitude with US 100 is accurate. But only work up to 60cm. Above 60cm, the value just came back to -1cm.
Any idea why this is happening?
Thank you,

DzikuVx, 2019-04-03:
On a quad? This is expected, all ultrasonic rangefinders suffer from working near propellers with all that noise and distrurbed air coming from them. Tops is like 1.5m above concrete but not always

minhdungDIY, 2019-04-03:
Hi,
Thanks for replying,
Yes, it's on a quad. But it shouldn't get any disturbance since I was just testing on computer (INAV configurator) without props on and the motors were at stop. I tried the Arduino sketch from the link bellow and it's showing higher than 60cm: 
https://github.com/trunglp/inav_sonar_i2c_arduino. 
This just makes me think that there's some bug in the Arduino sketch.
This problem also was mentioned at the end of this thread:
https://github.com/iNavFlight/inav/issues/2475
Let me know if you have any update on this.
Bests,

DzikuVx, 2019-04-04:
Arduino uses this:
```
#ifdef USE_US100
  #define PULSE_TO_CM 59 //Multiplier pulse length to distance in [cm]
  #define MAX_RANGE 300 //Range of 4 meters
#endif
```

So cutoff is at 3m, not 60cm. I will try to test it in upcoming weeks

georgekucher, 2019-04-29:
Hi guys,
@minhdungDIY thanks for mentioning.
@DzikuVx I've tried to test rangefinder on the fly but it seemed like it didn't work even on <60cm altitude. ANGLE, NAV ALTHOLD and SURFACE modes were enabled. And I was able to change altitude from the ground to 3+ meters. Is this expected behavior? Does the type of the surface matter and how much?
[Here](https://www.youtube.com/watch?v=cSqpAKiopB4) in Pawel's video everything goes smoothly.
I'll attach a log.
[LOG00139(cut).TXT](https://github.com/iNavFlight/inav-rangefinder/files/3126376/LOG00139.cut.TXT)
And [here](https://photos.app.goo.gl/53vH2DN8hag8QUYn6) is the video of this fly so that you see the surface.
Please help me with that.
Thanks in advance.

georgekucher, 2019-04-29:
BTW
What is this parameter used (or not) for?
[image]

georgekucher, 2019-06-02:
OK. I finally sorted our this sketch.
So. PULSE_TO_CM is not used for calculations of distance, at least for US 100. It is used (multiplied by MAX_RANGE) only for cutting off max range.
Another thing is that cutting off has a strange behavior. When MAX_RANGE is 300 (which is 3 meters not 4 how is written in cooment) it works only up to 60 cm. Vert weird for me. But if I set MAX_RANGE say 600 it works fine.

georgekucher, 2019-06-03:
Tested it yesterday. Max altitude in SURFACE mode was still 60-80. No. Even not like this. Whenever I switched into SURFACE mode, whatever altitude copter was at, it went down to 60-80 cm above the ground.
Unfortunately I had a crash that very flight (bad soldering at battery). And I can't open log files. "error opening log no logs in this file could be parsed successfully" it says.
If anyone knows how to fix it I would very much appreciate for the help.
Bag coincidences(((

englebert, 2020-12-08:
I think is the sonar sensor issue. I had bought many of these sonar modules, some works great and some just not.