iNavFlight/inav-rangefinder #8: INAV 2.2.0  RANGEFINDER=UNAVAILABLE

lugica2001, 2019-07-01:
INAV_I2C does not work on MATEKF722. 

I followed the suggestion in 
https://github.com/iNavFlight/inav-rangefinder/issues/4
bust RANGEFINDER is UNAVAILABLE

I checked with the oscilloscope and the FC does not send anything at the address 0x14.

richard0hidvegi, 2019-08-28:
> 
> 
> INAV_I2C does not work on MATEKF722.
> 
> I followed the suggestion in
> #4
> bust RANGEFINDER is UNAVAILABLE
> 
> I checked with the oscilloscope and the FC does not send anything at the address 0x14.

Hello,

I am working with a similar setup. I am using a 5V nano with level converters. I noticed a strange behavior on my system. If I power up the arduino first and after starting a main board the FC starting the communication. It might be just an isolated incident but maybe it can help. I checked the arduino part and it seems to be the problematic part for me.

DzikuVx, 2019-08-30:
Arduino has to boot before FC. Driver detection is performed only once and I have this device currently connected to Matek F722-STD and it works

closed 2019-08-30 by DzikuVx